SJSU-AD/FusionAD #110: Build error for covar package

menson168, 2018-11-15:
[image]

ntwong0, 2018-11-15:
Hmm a link error. If this error is coming up despite your having apt installed libgsl, then I suspect libgsl might not be in your $PATH variable.

Try
export PATH=/usr/lib:/usr/include/gsl:$PATH
catkin_make clean
catkin_make -j4 

If that still doesn't work, we might need to adjust CMakeLists.txt to include this at line 8
add_compile_options(-lgsl -lgslcblas -Iinclude)

I'll also try this when I'm back on my computer tomorrow

ntwong0, 2018-11-17:
### Summary:
After downloading a fresh Vagrant box with 16.04 LTS and installing ROS, I could not reproduce the problem listed above. However, I found and made corrections on a build error arising from building covar's services. I also made a correction to covar's README.md instructions regarding the `git clone` command - it now specifies the correct branch (ROSify) to use to build covar.

### The build error I encountered:

`{$ROS_WS}/src/covar/include/covar_ros.h:10:38: fatal error: covar/gen_pose_covar_srv.h: No such file or directory`

### Details: 
#### Setup and run the Vagrant box
0. Install Vagrant and VirtualBox.
    1. [Vagrant](https://www.vagrantup.com/docs/installation/)
    2. [VirtualBox](https://www.virtualbox.org/wiki/Downloads)
1. `mkdir my_vm_dir; cd my_vm_dir`
2. `vagrant init ubuntu/xenial64`
3. `vagrant ssh`

#### Setup the environment, install ROS
1. Set up swap file
    1. `cd /`
    2. `sudo dd if=/dev/zero of=swapfile bs=1M count=4000`
    3. `sudo mkswap swapfile`
    4. `sudo swapon swapfile`
    5. `sudo vim etc/fstab`
        1. Add the line `/swapfile none swap sw 0 0`
    6. `cat /proc/meminfo` to confirm swapfile is active
2. Install ROS via [these instructions](http://wiki.ros.org/kinetic/Installation/Ubuntu)
```
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
sudo apt-get update
sudo apt-get install ros-kinetic-desktop-full
sudo rosdep init
rosdep update
echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc
source ~/.bashrc
sudo apt-get install python-rosinstall python-rosinstall-generator python-wstool build-essential
```
#### Set up the ROS workspace, attempt to install covar, encounter the problem
1. Set up the ROS workspace
```
cd ~
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin_make
source devel/setup.bash
```
2. Attempt to install covar as instructed [here](https://github.com/ntwong0/test_gsl/blob/ROSify/README.md)
```
cd src
sudo apt-get update
sudo apt-get install libgsl2 libgsl-dev libgsl-dbg gsl-bin
git clone https://github.com/ntwong0/test_gsl.git -b ROSify covar
cd ..
catkin_make # The build error mentioned above results from this command
```
Although the build error arose, attempting `catkin_make` again resolved the issue. The fixes mentioned in the summary and implemented by commits (9be8930)[https://github.com/ntwong0/test_gsl/commit/9be89304c5edf2a015892bfcc83e7c51ee429a44] and (859d221)[https://github.com/ntwong0/test_gsl/commit/859d221dbd943f9a479a21fb907e64c15ea86ba7] patches this error.

menson168, 2018-11-25:
> ### Summary:
> After downloading a fresh Vagrant box with 16.04 LTS and installing ROS, I could not reproduce the problem listed above. However, I found and made corrections on a build error arising from building covar's services. I also made a correction to covar's README.md instructions regarding the `git clone` command - it now specifies the correct branch (ROSify) to use to build covar.
> 
> ### The build error I encountered:
> `{$ROS_WS}/src/covar/include/covar_ros.h:10:38: fatal error: covar/gen_pose_covar_srv.h: No such file or directory`
> 
> ### Details:
> #### Setup and run the Vagrant box
> 1. Install Vagrant and VirtualBox.
>    
>    1. [Vagrant](https://www.vagrantup.com/docs/installation/)
>    2. [VirtualBox](https://www.virtualbox.org/wiki/Downloads)
> 2. `mkdir my_vm_dir; cd my_vm_dir`
> 3. `vagrant init ubuntu/xenial64`
> 4. `vagrant ssh`
> 
> #### Setup the environment, install ROS
> 1. Set up swap file
>    
>    1. `cd /`
>    2. `sudo dd if=/dev/zero of=swapfile bs=1M count=4000`
>    3. `sudo mkswap swapfile`
>    4. `sudo swapon swapfile`
>    5. `sudo vim etc/fstab`
>       
>       1. Add the line `/swapfile none swap sw 0 0`
>    6. `cat /proc/meminfo` to confirm swapfile is active
> 2. Install ROS via [these instructions](http://wiki.ros.org/kinetic/Installation/Ubuntu)
> 
> ```
> sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
> sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
> sudo apt-get update
> sudo apt-get install ros-kinetic-desktop-full
> sudo rosdep init
> rosdep update
> echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc
> source ~/.bashrc
> sudo apt-get install python-rosinstall python-rosinstall-generator python-wstool build-essential
> ```
> #### Set up the ROS workspace, attempt to install covar, encounter the problem
> 1. Set up the ROS workspace
> 
> ```
> cd ~
> mkdir -p ~/catkin_ws/src
> cd ~/catkin_ws
> catkin_make
> source devel/setup.bash
> ```
> 1. Attempt to install covar as instructed [here](https://github.com/ntwong0/test_gsl/blob/ROSify/README.md)
> 
> ```
> cd src
> sudo apt-get update
> sudo apt-get install libgsl2 libgsl-dev libgsl-dbg gsl-bin
> git clone https://github.com/ntwong0/test_gsl.git -b ROSify covar
> cd ..
> catkin_make # The build error mentioned above results from this command
> ```
> Although the build error arose, attempting `catkin_make` again resolved the issue. The fixes mentioned in the summary and implemented by commits [9be8930](https://github.com/ntwong0/test_gsl/commit/9be89304c5edf2a015892bfcc83e7c51ee429a44) and [859d221](https://github.com/ntwong0/test_gsl/commit/859d221dbd943f9a479a21fb907e64c15ea86ba7) patches this error.

The mentioned build error can be fixed by running the command below before conducting ```catkin_make```:
```catkin_make covar_generate_messages_cpp```

closed 2018-11-25 by ntwong0