SJSU-AD/FusionAD #157: Refactor control

joeyzhu00, 2018-12-26:
Control should include the tf logic required to unpack the `Chassis_state` message into its quaternion components.

menson168, 2019-01-02:
Overall changes I will be making to the control node to adapt to the fully functional ```chassis_state``` message:

- Remove Subscriber for IMU node
- Remove all flags and trigger controlled by the IMU callback
- _MAYBE (Remove the transformation code that transforms the position from ```chassis_state``` to the front axle.)_

@joeyzhu00 Do I still need to transform the position from the ```chassis_state``` message to the front axle, or is the position given in the ```chassis_state``` already tracking the front axle?

joeyzhu00, 2019-01-02:
No, I forgot to add that in the original issue comment. Remove the transformation that is in the high-level control code.

PR #162 (2019-01-02): Iss 157
closes #157 

Performed all changes as mentioned in #157 .
closed 2019-01-02 by menson168