SJSU-AD/FusionAD #160: EKF Velocity Estimate Incorrect

joeyzhu00, 2019-01-01:
Use velocity output from wheel odometry for the `odom_to_chassis_state.py` program rather than the velocity output from the EKF.

joeyzhu00, 2019-01-01:
The figure below shows that there is a wind-up period for velocity estimates for the EKF and in general, the speed (magnitude) estimate is higher than that of the wheel odometry. There is also the problem of when the car starts moving in reverse, which results in underestimates in velocity. 

[image]

menson168, 2019-01-02:
> Use velocity output from wheel odometry for the `odom_to_chassis_state.py` program rather than the velocity output from the EKF.

Can you add an extra sentence on explaining why such change is needed? Just for clarification purpose. Thanks.

menson168, 2019-01-02:
> The figure below shows that there is a wind-up period for velocity estimates for the EKF and in general, the speed (magnitude) estimate is higher than that of the wheel odometry. There is also the [...]

Please elaborate a bit more on the wind-up, what does it mean and how does it affect the estimation. Do you mean that during the wind-up period, the velocity estimation from the EKF should match the output from the wheel odometry?

menson168, 2019-01-02:
> The change is needed because the purpose of `odom_to_chassis_state.py` is to publish a `Chassis_state` message that reflects the state of the vehicle. A bad velocity estimate being published would surely mess up the high-level controller.
> 
> The wind-up would not be favorable in terms of a control standpoint. The velocity estimation from the EKF should be similar to that of the wheel odometry, but it currently isn't.
> 
> I plotted X-Position versus Velocity because I wanted to show the difference in velocity estimates between the EKF and the wheel odometry over the course of travel with a reference to the current pose. This could have been plotted against time and have had the same result, but the key takeaway is the drop in velocity estimation when the car moves in the negative X-Position.

So does that mean the velocity output of ```odom_to_chassis_state.py``` will solely rely on the velocity input from wheel odometry? 

Can you point out where is the wind-up? I also cannot find which part of the graph depicts the vehicle is reversing.

What is the reason behind the blue data separating into two from x-position = 5 to x-position = 40?

joeyzhu00, 2019-01-02:
1) Yes
2) The wind-up starts around 5 [m]. The x-position reverses at around x = 37 [m].
3) The data is not separate, it's one whole set. Here's the pose graph for reference. 

[image]

closed 2019-01-02 by joeyzhu00

menson168, 2019-01-02:
> 1. Yes
> 2. The wind-up starts around 5 [m]. The x-position reverses at around x = 37 [m].
> 3. The data is not separate, it's one whole set. Here's the pose graph for reference.
> 
> [image]
I see that it was not separated, but during the windup, there is clearly a chunk of velocity estimation that lies in the neighborhood of the wheel odometry meanwhile the main trend of the velocity estimation is still way under the wheel odom. Why do you think that occurs? 
Also, the windup occurred in both the wheel odom and the estimation at around x = 5. Do you think the windup in the estimation is caused by the wheel odom?

joeyzhu00, 2019-01-02:
I don't have data for evidence for the first question, but I suspect that it may be from the settings in the `ekf_template.yaml` file. I could play around with those and see whether I can get `odom_to_chassis_state.py` to not have to subscribe to imu or wheel odometry. I'll create a separate issue for that.

I believe it's from the EKF, but I don't have a way to prove whether that is true. 

Currently, the only two contributors to the velocity estimation are the IMU and the wheel odometry.

menson168, 2019-01-02:
> I don't have data for evidence for the first question, but I suspect that it may be from the settings in the `ekf_template.yaml` file. I could play around with those and see whether I can get [...]

Interesting, I was curious at first to see that the estimation (blue) data suddenly drifted upward and created that cloud of data. Definitely, something funky was going on there lol.

PR #161 (2019-01-02): Iss 160
Closes #160