SJSU-AD/FusionAD #168: Migrate GPS Driver to ETHZ Piksi

menson168, 2019-01-05:
[https://github.com/ethz-asl/ethz_piksi_ros](ETHZ Piksi Driver)

menson168, 2019-01-05:
For @drofp and @joeyzhu00   

- Use ```/piksi/navsatfix_best-fix``` in place of ```/gps/fix``` from the old SwiftNav Driver  
  
There is also an ENU conversion built into the driver, it's probably worth it to compare it to our conversion. 

See [Optional ENU Conversion Setup](https://github.com/SJSU-AD/FusionAD/wiki/GPS-Setup---Swiftnav-Piksi-Multi#optional-enu-conversion-setup)

menson168, 2019-01-05:
See [GPS Setup](https://github.com/SJSU-AD/FusionAD/wiki/GPS-Setup---Swiftnav-Piksi-Multi) for the GPS setup wiki page

drofp, 2019-01-08:
Hand slipped^

menson168, 2019-01-14:
Closing issue.
Successfully replicate the migration process on Joey's laptop.

drofp, 2019-01-14:
*NOTE:* This has not been validated on my laptop yet. We may want to consider keeping this open until we have multiple laptops that can support this, or until Joey's laptop has been backed up on Clonezilla

menson168, 2019-01-14:
Reopen because @drofp need to try it out on his laptop.

menson168, 2019-01-23:
@drofp Please close this when you have validated the GPS.

closed 2019-01-23 by menson168

drofp, 2019-01-23:
Validated on most recent redcar test date

PR #169 (2019-01-06): Iss 168
Closes #168