SJSU-AD/FusionAD #186: Point Cloud to Laser Scan

joeyzhu00, 2019-01-15:
Point cloud to laser scan for 2D scan matching with [pointcloud_to_laserscan](http://wiki.ros.org/pointcloud_to_laserscan).

menson168, 2019-01-15:
Are we doing 2D mapping instead of 3D?

joeyzhu00, 2019-01-15:
There is also [cartographer](http://wiki.ros.org/cartographer) from Google which is a 3D approach.

menson168, 2019-01-15:
One thing that I am concerned with map building at the moment, specifically 2D maps, is the construction going on Campus right now. Since a lot of things are being moved around (LOL A.S. House), the lack of visual features provided by 2D maps might not work well for outdoors map-based localization. I think we should definitely try it with indoor though.

menson168, 2019-01-15:
> Gmapping seems to be pretty tempting because it is a SLAM based package and has the mapping component embedded. I'm not sure which package would be more accurate between AMCL and gmapping.

I did some googling around and seems like people have a better experience with AMCL than Gmapping. They said Gmapping's SLAM map isn't that stable and useable. I think AMCL might be the way to go, now just the matter of what to use to build that 2D map.

menson168, 2019-01-15:
The Cartographer package looks well supported. Since you had your eyes on the 2D mapping already, maybe try out the 2D approach first? Even if we can build an indoor map that would be dope!

joeyzhu00, 2019-01-15:
The ROS Navigation stack seems to endorse Gmapping http://wiki.ros.org/navigation/MapBuilding, but I think that we will be using AMCL + hector_mapping. I think that the performance outdoors may be hindered as you mentioned because of the lack of features but I think that this is still worth pursuing. We will be building a 3D map-based localization package as well, but the in-depth tutorials in these packages made it pretty tempting to get these working to at least have a comparison between our methods and the methods that are out there already.

joeyzhu00, 2019-01-15:
So it looks like the lidar driver actually has a /scan output. I didn't notice that until I started trying to get [pointcloud_to_laserscan](http://wiki.ros.org/pointcloud_to_laserscan) working (didn't output anything in its respective topic when i fed it a bag of lidar data). The lidar driver's method is to just read off of a single scan line. Do you guys think that it would be appropriate to just use the built-in scan feature from the velodyne driver or should we look for something that accounts for all of the point cloud data to generate a scan message?

drofp, 2019-01-15:
@joeyzhu00 the `pointcloud_to_laserscan` might be looking for a specific topic name that isn't currently publishing.  

I think it would be safer to try to get it to work to get a more accurate representation of the 3D data, rather than relying on a single scanline. If this takes more than a few hours, however, probably just put it on hold for now and use the builtin `/scan`

ntwong0, 2019-01-15:
Hated trying to get cartographer to work when I started using ROS two years ago, but now that I've learned how to do message sync I wouldn't mind tackling this again

I ended up using hector mapping for localization, I used the odom  it produced to fuse into robot_localization

Would be interesting to compare

 the pointcloud from velodyne  vs
 scan from velodyne vs 
pointcloud to laser scan

closed 2019-03-28 by joeyzhu00