SJSU-AD/FusionAD #244: [Control - DBW] DBW Steering Disengage Method

menson168, 2019-04-07:
There should be a seamless way to disengage the autonomy mode and recover. @joeyzhu00 and @nathansrinivasan5 please work together with me on brainstorming a solution for this issue.

joeyzhu00, 2019-04-07:
We can experiment with using a button to engage and disengage the steering. I can add logic to stop sending CAN messages if the disengaged status is published as true.

joeyzhu00, 2019-04-07:
We can also experiment with the built-in torque-sensing capabilities of the EPAS motor, but there is some noise associated with it that have to be addressed (false positives from the output of the motor).

menson168, 2019-04-07:
> We can experiment with using a button to engage and disengage the steering. I can add logic to stop sending CAN messages if the disengaged status is published as true.

> We can also experiment with the built-in torque-sensing capabilities of the EPAS motor, but there is some noise associated with it that have to be addressed (false positives from the output of the motor).

I think the button/switch approach is desirable in the consideration of deployment time. Will probably require less analysis than studying how to not run into false-positives on the torque sensor.

nathansrinivasan5, 2019-04-07:
In the scenarios we have faced before when we needed to disengage the steering, do we have to know where the next nearest waypoint is when engaging it again?

drofp, 2019-04-07:
> In the scenarios we have faced before when we needed to disengage the steering, do we have to know where the next nearest waypoint is when engaging it again?

Just check in Rviz if we have to keep going afterwards

menson168, 2019-04-08:
> > In the scenarios we have faced before when we needed to disengage the steering, do we have to know where the next nearest waypoint is when engaging it again?
> 
> Just check in Rviz if we have to keep going afterwards

The re-engage position can also be just an approximately nearby location with respect to the path. The controller would notify in error if you are too far off the path. (Donut Selector)